rodrigo-xavier/GuitarHero #27: Módulo Debug

rodrigo-xavier, 2019-04-02:
Tem sido necessário debugar o projeto à fim de encontrar erros e bugs de código, assim sendo, crie um módulo debug que permite escolher que tipo de debug deve ser feito.

cesaraugc, 2019-04-25:
Além disso, colocar todos os pixels que estão sendo utilizados neste módulo. Ainda estão apenas alguns dos pixels.

rodrigo-xavier, 2019-04-25:
Incluir também uma forma de impedir o matlab de enviar sinal de pressionar para o arduino.

rodrigo-xavier, 2019-05-16:
Módulo debug deve poder escolher qual nota pressionar. Sendo ela uma branch separada da master.

rodrigo-xavier, 2019-05-23:
Os códigos da branch debug devem poder ser facilmente atualizados com a master através do merge, sem que hajam conflitos.

cesaraugc, 2019-06-06:
Acredito que o debug possa ser incluído na master, já que também há a opção de jogar em todas as cores.