rodrigo-xavier/GuitarHero #33: Atualizar função de pegar tempo

rodrigo-xavier, 2019-04-26:
O objetivo é obter um tempo mais preciso .

Para isso, é necessário pegar o tempo da primeira nota, verificar se não não é detectada nenhuma nota/rastro em qualquer das trilhas num tempo equivalente e maior que o da primeira nota, e iniciar a verificação de tempos para aquela trilha específica. Ao final de tudo realizar uma média dos tempos capturados.

cesaraugc, 2019-06-14:
Corrigido?

rodrigo-xavier, 2019-06-14:
Remover bug da variável temp

rodrigo-xavier, 2019-07-12:
Branch detect_times_upgrade2

- Implementar um algoritmo de detecção de notas mais sofisticado (O detect_times pode estar realizando mais tics do que tocs e isso quebra o algoritmo)
- (Futuro) Implementar uma forma de pegar  tempo a partir da metade da música